apfaudio/eurorack-pmod #7: Faust support?

dromer, 2023-02-09:
Maybe a bit of an odd question. Since Faust can be compiled to run on FPGA hardware, would it be possible to target your board as well?

https://fast.grame.fr/results/processor.html

schnommus, 2023-02-11:
Hey there @dromer,

Indeed I have seen this project and it looks super interesting.

From what I read it currently depends on the proprietary Xilinx/Vivado HLS toolchain to translate from generated cpp (from the Faust compiler) into HDL, and this HLS step is not open source.

However looking at this presentation on the topic: https://ccrma.stanford.edu/~rmichon/talks/ccrma-openhouse-21/#/4

It seems to suggest that they are actively working on a fixed-point target for Faust as well as a faust2vhdl target which would theoretically be able to translate Faust DSP graphs into HDL without requiring an external HLS tool. That would allow one to target the open-source FPGA toolchain. I'm not sure what the status of that project is, but if that is up and running one day, it would be pretty easy to integrate with the work on this project.

A completely different approach (which is possible using LiteX or similar) could be to instantiate one or more RiscV cores on the FPGA and run compiled Faust on those. It would require a bit of plumbing, but one could then flexibly decide which parts of the DSP run on a 'faust softcore', and which parts are implemented in gateware. In fact I started working on an example of exactly this, but it might not be ready for a little while :)

dromer, 2023-02-11:
Cool! thnx for looking into this @schnommus and for the elaborate answer :)

closed 2023-07-01 by schnommus