apfaudio/eurorack-pmod #20: Support builds on Windows + Mac

schnommus, 2023-04-07:
Currently only builds on Linux are tested, however it should be relatively trivial to support other operating systems.

I just tried installing oss-cad-suite and MSYS2 on a Windows VM and was able to build the examples in this repository fine, with the following notes:
- Must change all relative paths of .hex files such that they are relative to top-level makefile
- Do all building under MSYS2 MINGW64 shell, adding oss-cad-suite bin/lib directories to PATH
- After doing this, 'make' seems to work fine

Ideally we should add Windows/Mac targets to CI as well so they stay healthy

schnommus, 2023-04-13:
As of https://github.com/apfelaudio/eurorack-pmod/pull/24 windows builds are now supported and in CI.

However it might be worth making the linter/test bench CI jobs also run on windows

schnommus, 2023-07-01:
I don't physically have any Apple hardware so it's difficult for me to add Mac support.

I'll leave this open in case someone else feels like taking it up

forrestbaer, 2023-12-21:
I took a quick go at this on my `M1 24" IMac running Sonoma 14.1.1`, and was able to run `make BOARD=icebreaker CORE=filter` under the `gateware` folder successfully and get four build files `top.asc`, `top.bin`, `top.json`, `top.log`.

Running `make` by itself asked for the BOARD definition at minimum.

As for the rest I don't know! I'm considering manufacturing a few of the current 3.3 designs to play with, very cool project.

forrestbaer, 2023-12-21:
Would be happy to help you close this one, if there are other tests I can do let me know.

schnommus, 2023-12-23:
Hey there @forrestbaer, thanks for testing this on Mac! Good to hear that it wasn't too hard to get things running :)
I wanted to keep this ticket open until I have a Mac target added to the CI workflow in the `.github` folder, at the moment we automatically check the build succeeds on linux & windows after every push.

As for the hardware, if you can wait until Jan 31 or so, I already have a new batch of 50 assembled+tested boards ready to ship out - the delay is just because I am travelling all of January and can't ship things during that time :). In any case if you can't wait, feel free to manufacture them yourself (just keep in mind it is more difficult for me to support anyone using hardware that I have not tested myself beforehand!)

forrestbaer, 2023-12-23:
Awesome! I hope to nab a pair before they're gone.

PR #52 (2024-01-16): ci: add automated macos build to ci
Closes #20
closed 2024-01-16 by schnommus