apfaudio/eurorack-pmod #26: add support for ICE-V-Wireless

proppy, 2023-04-18:
what would be needed to add support for https://github.com/ICE-V-Wireless/ICE-V-Wireless ?

would changing the [pinmap](https://github.com/apfelaudio/eurorack-pmod/blob/master/gateware/boards/icebreaker/pinmap.pcf) be enough?

looking at https://github.com/ICE-V-Wireless/ICE-V-Wireless/blob/main/Gateware/blinky_demo/icestorm/Makefile#L11-L12 it seems based on the same part than the icebreaker.

ps: thanks for the pictures of berlin!

schnommus, 2023-04-18:
Hey there @proppy! glad you liked the pictures :)

On a first glance at that board, you should indeed be fine with just a pinmap change. The chip is the same, the external clock to the FPGA is also 12MHz like the icebreaker, and the PMODs power outputs are by default driven at 3.3V which is correct.

It might be worth cross-referencing your pinmap change against the physical diagram of the icebreaker pinout to make sure the changed mapping physically looks the same (current assignment in this repo is PMOD 2): https://www.crowdsupply.com/img/e8c5/icebreaker-v1-0b-legend.jpg

Let me know how you go -- if it works let's merge it in.

schnommus, 2023-07-01:
Closing this for now as I don't actually have the hardware in question, but if you want to add support for this board I'd definitely review your PR

closed 2023-07-01 by schnommus