apfaudio/eurorack-pmod #30: power fpga board from the 5v eurorack connector

proppy, 2023-06-30:
I noticed that the 5v power line on the eurorack connector is grounded, is that intentional?

I'd love to be able to power the dev board directly from the eurorack supply? (or is a #badidea?)

schnommus, 2023-07-01:
Hey there @proppy,

The board only has a 10-pin eurorack power connector, [not a 16-pin one](https://doepfer.de/a100_man/a100b_e.gif). 10-pin eurorack power cables don't actually have a 5V line, so I'm not sure what you mean by the 5V line being grounded. Or do you mean it's grounded in the cable itself which was shipped? That would be a quality issue!

That being said, I intentionally put enough space next to the power connector such that you could in theory use a 16-pin to 16-pin IDC cable instead of a 16-pin to 10-pin IDC cable (as shipped). Given the 5V ingress is one pin over from the +12V input, you could fit a single [2.54" pin header cable ](https://m.media-amazon.com/images/I/41lQWFOiQnL._AC_UF894,1000_QL80_.jpg) in there to connect your board. In this arrangement you can power everything just from 1x16-pin power cable with the existing hardware. Like this (using a proper cable instead of the one pictured here that is only crimped on the first 10 pins):

[image]

As for powering boards directly from the PMOD connector, this will not work as there is no 3V3 regulator actually on the PMOD. I considered adding an optional jumper for this feature actually [in the next hardware release](https://github.com/apfelaudio/eurorack-pmod/issues/21), but I'm not so sure about this feature as a few boards I have played with would not be happy with this arrangement, and I guess most PMODs on the market expect to be powered by the board, not the other way around.

schnommus, 2024-02-04:
The current hardware release (R3.3) added an additional 'backpack' FFC connector on the back which can be used to bolt on daughterboards to the eurorack-pmod with 1 connector such that they can A) siphon off power and B) send audio/etc to the PMOD with a shared (optional) connector

This enables the use-case where we want a dev-board to be powered by the PMOD board in a neat way, given an adapter PCB which holds the dev board in a nice orientation for eurorack.

Such an adapter PCB doesn't exist yet, but at least it's possible now :)

closed 2024-02-04 by schnommus