apfaudio/eurorack-pmod #43: make calls icepack but not iceprog?

matpalm, 2023-10-14:
```
git clone git@github.com:apfelaudio/eurorack-pmod.git
# commit ec676d6eed92159870c07cd1b16ddc0b48d029ef
cd eurorack-pmod
git submodule update --init --recursive
cd gateware
make BOARD=icebreaker CORE=mirror
```
runs yosys, nextpnr and icepack, but not the final iceprog?

```
mkdir -p build/icebreaker
# For now we always force a re-build since we can pass different DSP cores
# through environment vars and we need a re-build to happen in this case.
make -B -f boards/icebreaker/Makefile BUILD=build/icebreaker CORE=mirror 
make[1]: Entering directory '/home/mat/dev/eurorack-pmod/gateware'
yosys -f "verilog -sv "-DSELECTED_DSP_CORE=mirror -DINVERT_BUTTON=1 -DICE40"" -ql build/icebreaker/top.log -p 'synth_ice40 -dsp -top top -json build/icebreaker/top.json' top.sv boards/icebreaker/sysmgr.v eurorack_pmod.sv drivers/pmod_i2c_master.sv drivers/ak4619.sv external/no2misc/rtl/uart_tx.v external/no2misc/rtl/i2c_master.v cal/cal.sv cal/debug_uart.sv cores/mirror.sv cores/clkdiv.sv cores/seqswitch.sv cores/sampler.sv cores/bitcrush.sv cores/vca.sv cores/vco.sv cores/pitch_shift.sv cores/stereo_echo.sv cores/filter.sv cores/util/filter/karlsen_lpf_pipelined.sv cores/util/filter/karlsen_lpf.sv cores/util/transpose.sv cores/util/echo.sv cores/util/delayline.sv cores/util/dc_block.sv cores/util/wavetable_osc.sv
Warning: Yosys has only limited support for tri-state logic at the moment. (top.sv:126)
Warning: Yosys has only limited support for tri-state logic at the moment. (top.sv:127)
Warning: Resizing cell port $paramod\debug_uart\DIV=s32'00000000000000000000000000001100.utx.div from 32 bits to 8 bits.
Warning: Resizing cell port top.eurorack_pmod1.force_dac_output from 32 bits to 16 bits.
nextpnr-ice40 --up5k \
--package sg48 \
--json build/icebreaker/top.json \
--pcf boards/icebreaker/pinmap.pcf \
--asc build/icebreaker/top.asc \

_<snipped lots of nextpnr output >_

Info: Max delay <async>                     -> posedge PMOD_MCLK$SB_IO_OUT: 14.85 ns
Info: Max delay posedge $PACKER_GND_NET     -> posedge PMOD_MCLK$SB_IO_OUT: 16.14 ns
Info: Max delay posedge PMOD_MCLK$SB_IO_OUT -> <async>                    : 10.75 ns
Info: Max delay posedge PMOD_MCLK$SB_IO_OUT -> posedge $PACKER_GND_NET    : 16.14 ns

Info: Program finished normally.
icepack build/icebreaker/top.asc build/icebreaker/top.bin
make[1]: Leaving directory '/home/mat/dev/eurorack-pmod/gateware'
```

but can run iceprog explicitly ok?

```
$ iceprog build/icebreaker/top.bin
init..
cdone: high
reset..
cdone: low
flash ID: 0xEF 0x40 0x18 0x00
file size: 104090
erase 64kB sector at 0x000000..
erase 64kB sector at 0x010000..
programming..
done.                 
reading..
VERIFY OK             
cdone: high
Bye.
```

it's in the Makefile, but isn't run?

matpalm, 2023-10-14:
actually, same for the ecpix5...

```
make BOARD=ecpix5 CORE=mirror
```

finished with 

```
...
mkdir -p build/icebreaker
# For now we always force a re-build since we can pass different DSP cores
# through environment vars and we need a re-build to happen in this case.
make -B -f boards/icebreaker/Makefile BUILD=build/icebreaker CORE=mirror 
make[1]: Entering directory '/home/mat/dev/eurorack-pmod/gateware'
yosys -f "verilog -sv "-DSELECTED_DSP_CORE=mirror -DINVERT_BUTTON=1 -DICE40"" -ql build/icebreaker/top.log -p 'synth_ice40 -dsp -top top -json build/icebreaker/top.json' top.sv boards/icebreaker/sysmgr.v eurorack_pmod.sv drivers/pmod_i2c_master.sv drivers/ak4619.sv external/no2misc/rtl/uart_tx.v external/no2misc/rtl/i2c_master.v cal/cal.sv cal/debug_uart.sv cores/mirror.sv cores/clkdiv.sv cores/seqswitch.sv cores/sampler.sv cores/bitcrush.sv cores/vca.sv cores/vco.sv cores/pitch_shift.sv cores/stereo_echo.sv cores/filter.sv cores/util/filter/karlsen_lpf_pipelined.sv cores/util/filter/karlsen_lpf.sv cores/util/transpose.sv cores/util/echo.sv cores/util/delayline.sv cores/util/dc_block.sv cores/util/wavetable_osc.sv
Warning: Yosys has only limited support for tri-state logic at the moment. (top.sv:126)
Warning: Yosys has only limited support for tri-state logic at the moment. (top.sv:127)
Warning: Resizing cell port $paramod\debug_uart\DIV=s32'00000000000000000000000000001100.utx.div from 32 bits to 8 bits.
Warning: Resizing cell port top.eurorack_pmod1.force_dac_output from 32 bits to 16 bits.
nextpnr-ice40 --up5k \
--package sg48 \
--json build/icebreaker/top.json \
--pcf boards/icebreaker/pinmap.pcf \
--asc build/icebreaker/top.asc \

Info: constrained 'CLK' to bel 'X12/Y31/io1'
Info: constraining clock net 'CLK' to 12.00 MHz
Info: constrained 'UART_TX' to bel 'X15/Y0/io0'
Info: constrained 'RESET_BUTTON' to bel 'X16/Y0/io0'
Info: constrained 'PMOD_SDIN1' to bel 'X18/Y31/io1'
Info: constrained 'PMOD_SDOUT1' to bel 'X19/Y31/io1'
Info: constrained 'PMOD_LRCK' to bel 'X18/Y0/io1'
Info: constrained 'PMOD_BICK' to bel 'X21/Y0/io1'
Info: constrained 'PMOD_I2C_SCL' to bel 'X18/Y31/io0'
Info: constrained 'PMOD_I2C_SDA' to bel 'X19/Y31/io0'
Info: constrained 'PMOD_PDN' to bel 'X19/Y0/io1'
Info: constrained 'PMOD_MCLK' to bel 'X22/Y0/io1'

Info: Packing constants..
Info: Packing IOs..
Info: Packing LUT-FFs..
Info:      967 LCs used as LUT4 only
Info:      397 LCs used as LUT4 and DFF
Info: Packing non-LUT FFs..
Info:      386 LCs used as DFF only
Info: Packing carries..
Info:       87 LCs used as CARRY only
Info: Packing indirect carry+LUT pairs...
Info:       55 LUTs merged into carry LCs
Info: Packing RAMs..
Info: Placing PLLs..
Info:   constrained PLL 'sysmgr_instance.pll_I' to X12/Y31/pll_3
Info: Packing special functions..
Info: Packing PLLs..
Info:     Input frequency of PLL 'sysmgr_instance.pll_I' is constrained to 12.0 MHz
Info:     VCO frequency of PLL 'sysmgr_instance.pll_I' is constrained to 768.0 MHz
Info:     Derived frequency constraint of 24.0 MHz for net sysmgr_instance.clk_2x_i
Info:     Derived frequency constraint of 12.0 MHz for net PMOD_MCLK$SB_IO_OUT
Info:   PLL 'sysmgr_instance.pll_I' has LOCK output, need to pass all outputs via LUT
Info:   constrained 'sysmgr_instance.pll_lock_SB_LUT4_I3_LC' to X1/Y30/lc0
Info: Promoting globals..
Info: promoting eurorack_pmod1.cal_instance.out0_SB_DFFESR_Q_R [reset] (fanout 16)
Info: promoting eurorack_pmod1.cal_instance.out1_SB_DFFESR_Q_R [reset] (fanout 16)
Info: promoting eurorack_pmod1.ak4619_instance.last_fs_SB_LUT4_I2_O[0] [cen] (fanout 128)
Info: promoting eurorack_pmod1.cal_instance.state_SB_DFF_Q_D_SB_LUT4_O_I3_SB_LUT4_O_1_I1_SB_DFF_D_Q_SB_LUT4_I1_1_O[1] [cen] (fanout 128)
Info: promoting eurorack_pmod1.cal_instance.l_clk_fs_SB_LUT4_I2_O_SB_LUT4_I2_O [cen] (fanout 32)
Info: promoting eurorack_pmod1.cal_instance.l_clk_fs_SB_LUT4_I2_O_SB_LUT4_O_1_I3_SB_LUT4_I1_O [cen] (fanout 32)
Info: Constraining chains...
Info:       31 LCs used to legalise carry chains.
[... 393 lines cut ...]
Info:  3.0 16.0    Net eurorack_pmod1.cal_instance.out[0]_SB_DFFE_Q_1_D_SB_LUT4_O_I2_SB_LUT4_O_I3[2] (2,18) -> (0,15)
Info:                Sink eurorack_pmod1.cal_instance.out[0]_SB_DFFE_Q_30_D_SB_LUT4_O_I0_SB_LUT4_O_1_I2_SB_LUT4_O_I1_SB_MAC16_O_1_DSP.A_14
Info:                Defined in:
Info:                  top.sv:135.3-173.2
Info:                  cal/cal.sv:101.26-101.42
Info:                  eurorack_pmod.sv:78.2-101.2
Info:                  /opt/oss-cad-suite/lib/../share/yosys/ice40/arith_map.v:31.22-31.23
Info:  0.1 16.1  Setup eurorack_pmod1.cal_instance.out[0]_SB_DFFE_Q_30_D_SB_LUT4_O_I0_SB_LUT4_O_1_I2_SB_LUT4_O_I1_SB_MAC16_O_1_DSP.A_14
Info: 4.8 ns logic, 11.4 ns routing

Info: Max frequency for clock 'PMOD_MCLK$SB_IO_OUT': 20.01 MHz (PASS at 12.00 MHz)

Info: Clock '$PACKER_GND_NET' has no interior paths

Info: Max delay <async>                     -> posedge PMOD_MCLK$SB_IO_OUT: 14.85 ns
Info: Max delay posedge $PACKER_GND_NET     -> posedge PMOD_MCLK$SB_IO_OUT: 16.14 ns
Info: Max delay posedge PMOD_MCLK$SB_IO_OUT -> <async>                    : 10.75 ns
Info: Max delay posedge PMOD_MCLK$SB_IO_OUT -> posedge $PACKER_GND_NET    : 16.14 ns

Info: Program finished normally.
icepack build/icebreaker/top.asc build/icebreaker/top.bin
make[1]: Leaving directory '/home/mat/dev/eurorack-pmod/gateware'
```

but the final `openFPGALoader` isn't called?

matpalm, 2023-10-14:
looks like `icepack` process is running and exiting ok ( so it's not some kind of early exit from ther Make process? )

```
$ icepack -v build/icebreaker/top.asc build/icebreaker/top.bin
Parsing ascii file..
Writing bitstream file..
netpbm
fill done
Done.

$ echo $?
0

```

schnommus, 2023-10-14:
Thanks for the detailed log of the commands you used!

Currently the way the makefile is set up, the following command --

`$ make BOARD=icebreaker CORE=mirror`

Will only build the bitstream. If you wish to also program the board AFTER building the bitstream in one step, then you can append the `prog` target.

`$ make BOARD=icebreaker CORE=mirror prog`

matpalm, 2023-10-14:
Ah! Missed that, thanks!

closed 2023-10-14 by matpalm