apfaudio/eurorack-pmod #44: unsure if my eurorack pmod is running at all?

matpalm, 2023-10-14:
flashed my first firmware to an icebreaker ( with the caveat that i had to call `iceprog` explicitly ( see https://github.com/apfelaudio/eurorack-pmod/issues/43 ))

i think the flashing is successful?

```
$ iceprog build/icebreaker/top.bin
init..
cdone: high
reset..
cdone: low
flash ID: 0xEF 0x40 0x18 0x00
file size: 104090
erase 64kB sector at 0x000000..
erase 64kB sector at 0x010000..
programming..
done.                 
reading..
VERIFY OK             
cdone: high
Bye.
```

but am just seeing -10V from the all outputs..

setup is the following

dixie -> mordax ( green trace, sine wave ) -> eurorack pmod input 1 
eurorack pmod output 1 -> mordax ( blue trace, -10V) 

[image]

matpalm, 2023-10-14:
in fact, should i see LED activity on the eurorack pmod? none of them are lit...

schnommus, 2023-10-14:
Hey there @matpalm - indeed you should definitely see LED activity! I have intentionally (and unintentionally) connected both connectors in the wrong orientation on other eurorack-pmods without frying anything, so it should be pretty hard to fry these boards. (the +/- 12V input is reverse polarity protected, and the 3V3/PMOD side will just not do anything if connected backwards).

Some things to try:
0) Make sure you measure 3V3/GND across the marked PMOD pins (pinout on back of PCB). Although I am sure you already did that :).
1) For Icebreaker <=> PMOD, connect the ribbon cable connected such that the red stripe is on pin 1 on both the icebreaker & PMOD (this is the square pad on the OTHER side of the 3V3 line, which is SDI on the PMOD and [pin 27 on the icebreaker PMOD 2](https://www.crowdsupply.com/img/e8c5/icebreaker-v1-0b-legend.jpg)). Actually the way you have it connected should work anyway, but maybe best if you use the same orientation I had when testing the unit here.
2) Make sure you have pulled the latest version of this repository, I pushed some fixes about a week ago for a little timing bug (everything will work without the fix, but best to re-fetch everything!)
3) As a first sanity check, you can try even without the +/-12V connected. If you flash the mirror example, the LEDs will all be off. If you plug in a jack to input 1/alpha, you should see the LED light up solid as the jack detect IC sees something is connected (but there will be no output as the analog stages are all unpowered)
4) If you have an oscilloscope on hand maybe it's worth making sure you have traffic from the icebreaker on the SDA/SCL
5) I test every eurorack-pmod before shipping it, but I do NOT test the ribbon cables myself, maybe it's worth doing a pin-to-pin-check with a multimeter on that to make sure it's not a bad one? I have never seen a bad one but there's a first time for everything
6) If things still don't work, perhaps try the same example with the ECPIX5 (noting the ECPIX5 has pin1 marked on the OTHER side of its PMOD connector, so you can't go Pin1<->Pin1 like the icebreaker. Just make sure 3V3 goes to 3V3.

In any case I hope the above help you, I'm currently on holidays up in Norway so my answers may have a few days delay, but in any case rest assured if we can't get your unit to run I'm more than happy to send you another or issue a refund.

Cheers,
Seb.

matpalm, 2023-10-14:
awesome thank you!

for either the icebreaker or the ecpix5; if i connect just the PMOD connector, but not the eurorack power, i measure 3.3V on the PCB between GND and 3.3V . so that's good.

[image]

for the icebreaker i don't get any lit LEDs. 
but the ecpix5 i _do_ get them lit.
furthermore for the ecpix5, if i conenct the eurorack power the mirror core works now! 

so there's something wrong with my icebreaker possibly? i've flashed some other hello world button/LED firmwares to it, and they've all been fine, so maybe i botched the soldering of the headers to the PMOD2 ?

so, that's good news! for my use case the icebreaker was going to be too small anyways.

last week i ported the code for my neural net and it all runs ok in simulation ( though the code is a bit of a mess regarding port arrays etc https://github.com/matpalm/cached_dilated_causal_convolutions/blob/master/sverilog_version/src/network.sv  )

tomorrow i'll try to bring it into the eurorack-pmod code base and get it running. ( there is still some things to do re: input scaling, in terms of fixed point config, but it's very close now )

i'd be happy to say this is resolved, because i want to focus on the ecpix5, but at the same time if there's anything else i can do to debug the icebreaker to prove it's _not_ the eurorack-pmod please let me know, happy to help shake anything that might be going on.

as an aside, though it's nothing to do with this project directly, because i have a newish R3 ecpix5 i had to flash the ecp5 with a slightly different command since it has a newer FDTI chip than openFPGALoader expects. ( specifically i have to run  `openFPGALoader -c ft4232 build/ecpix5/top.bin` instead of `openFPGALoader -b ecpix5 build/ecpix5/top.bin`  see https://github.com/trabucayre/openFPGALoader/issues/377 for more info )

matpalm, 2023-10-14:
resolving because i'm sure this a problem with my icebreaker.. please reopen seb if you feel it's worth diving deeper

closed 2023-10-14 by matpalm

schnommus, 2023-10-15:
Hey there @matpalm, great to hear you got things working on the ECPIX5. Indeed given your symptoms I would suspect something is up with the iCEbreaker -- one thing that might be worth trying is using a slow upcounter and assign each bit to one of the pins on PMOD2 to check if every pin is connected or shorted somewhere.

In any case I will try the iCEbreaker examples again as a sanity check when I have access to the hardware, however the icebreaker is generally the platform on which I test everything first so it would surprise me if a bug has creeped into the gateware there.

schnommus, 2023-10-25:
For completeness on this ticket - I tested the examples on iCEbreaker again at `master` and things seem to be working fine so indeed I would suspect dud hardware