apfaudio/eurorack-pmod #50: Hardware R3.2 bringup: changelist + notes

schnommus, 2023-11-12:
proto1 looks like this:

[image]

I will submit a production batch for R3.3 likely in the next month or 2.

## Design / Gateware bringup
- Working gateware: this repo, branch `seb/r32-bringup`
- R3.2 hardware design: this repo, branch `seb/capsense-groundfix`
- Gateware for proximity/touch-sensitive jacks! (experimental) this repo, branch `seb/touch-fsm`

# R3.2 Hardware Changelist
- **feature: smaller / skiff friendly form factor**: total module depth (to the end of the pin headers) is drastically reduced from 45mm to about 32mm. This fits in a lot more skiff/thin eurorack cases.
- **feature: proximity + touch sensitive jacks**: any unused jacks can be used as proximity/touch sensors as an extra input method (I think this has never been done in Eurorack - but I always wanted to see if it's possible..). They are continuously sensitive from ~5mm away from the jack and emit continuous values 0-255. This increases audio noise a little bit, but the feature is switched off by default in the first branch above. Individual channels can also be dynamically switched on-off.
- **feature: auxiliary flat-flex connector on bottom-side**: all power nets and PMOD pins are broken out on an auxiliary connector underneath the board, this is intended for future daughterboards containing an FPGA/host to bolt onto the PMOD and extract power from the PMOD power connector.
- **feature: internal post-scaler loopback**: whenever both in/out jacks on a particular channel are not connected, inputs are automatically connected to outputs. This simplifies self-calibration, but I don't like how it affects noise performance, so I might remove this for the next revision.
- **bugfix**: scaler network adjustments. Previously eurorack-pmods were about 1V out from zero without calibration. Now all PMODs should be statistically close to zero even without calibration (but still require it for mV-level accuracy). This also buys us more in/out swing range, and because I also reduced the scaler amplitudes, improves noise performance.
- **bugfix**: fix routing error on star ground: this required manual rework on R3.1 and is no longer a problem. This also improves audio noise performance.

matpalm, 2023-11-12:
"proximity + touch sensitive jacks" wild!

schnommus, 2023-12-23:
Update: R3.3 hardware fixes the issues outlined above and tests A-OK. Noise performance is much better than all previous revisions and the nominal DC accuracy is much better than previous revisions without any calibration. The touch sensing also does not seem to interfere with the noise performance any more.

I have a new batch of assembled + tested R3.3 boards ready to ship at the end of January (I am travelling all of January and so can't ship things until then, that's the only reason things are delayed).

schnommus, 2024-02-04:
In the R3.3 production batch, after mass testing I have 4 failures of 50 units, mostly soldering issues from the manufacturer. In any case R3.3 is now in stock, so I am closing this :)